ZHomeSlice/Simple_MPU6050 #1: Compilation errors

AlexFundorin, 2019-08-28:
Why do I get lots of compilation errors like this:
```
lib/Simple_MPU6050/DMP_Image.h:530:27: error: ISO C++ forbids declaration of 'normalize' with no type [-fpermissive]
   Quaternion :: normalize() {
lib/Simple_MPU6050/DMP_Image.h:530:3: error: extra qualification 'Quaternion::' on member 'normalize' [-fpermissive]
   Quaternion :: normalize() {
lib/Simple_MPU6050/DMP_Image.h: In member function 'Quaternion Quaternion::setQuaternion(float*)':
lib/Simple_MPU6050/DMP_Image.h:496:3: warning: no return statement in function returning non-void [-Wreturn-type]
   }
   ^
lib/Simple_MPU6050/DMP_Image.h: In member function 'int Quaternion::normalize()':
lib/Simple_MPU6050/DMP_Image.h:536:12: error: invalid conversion from 'Quaternion*' to 'int' [-fpermissive]
     return this;
            ^
lib\Simple_MPU6050\Simple_MPU6050.cpp: In member function 'bool Simple_MPU6050::view_DMP_firmware_Instance(uint16_t)':
lib\Simple_MPU6050\Simple_MPU6050.cpp:574:36: error: invalid conversion from 'uint16_t {aka short unsigned int}' to 'uint16_t* {aka short unsigned int*}' [-fpermissive]
    writeWords(devAddr, 0x6D, 1,  ii);
```

williamesp2015, 2021-08-11:
A lot of errors in the DMP_image.h and lib I just removed some:
#ifndef DMP_Image_h
#define DMP_Image_h
#include <Arduino.h>
/* These defines are copied from dmpDefaultMPU6050.c in the general MPL
   releases. These defines may change for each DMP image, so be sure to modify
   these values when switching to a new image.
*/
#define CFG_LP_QUAT             (2712)
#define END_ORIENT_TEMP         (1866)
#define CFG_27                  (2742)
#define CFG_20                  (2224)
#define CFG_23                  (2745)
#define CFG_FIFO_ON_EVENT       (2690)
#define END_PREDICTION_UPDATE   (1761)
#define CGNOTICE_INTR           (2620)
#define X_GRT_Y_TMP             (1358)
#define CFG_DR_INT              (1029)
#define CFG_AUTH                (1035)
#define UPDATE_PROP_ROT         (1835)
#define END_COMPARE_Y_X_TMP2    (1455)
#define SKIP_X_GRT_Y_TMP        (1359)
#define SKIP_END_COMPARE        (1435)
#define FCFG_3                  (1088)
#define FCFG_2                  (1066)
#define FCFG_1                  (1062)
#define END_COMPARE_Y_X_TMP3    (1434)
#define FCFG_7                  (1073)
#define FCFG_6                  (1106)
#define FLAT_STATE_END          (1713)
#define SWING_END_4             (1616)
#define SWING_END_2             (1565)
#define SWING_END_3             (1587)
#define SWING_END_1             (1550)
#define CFG_8                   (2718)
#define CFG_15                  (2727)
#define CFG_16                  (2746)
#define CFG_EXT_GYRO_BIAS       (1189)
#define END_COMPARE_Y_X_TMP     (1407)
#define DO_NOT_UPDATE_PROP_ROT  (1839)
#define CFG_7                   (1205)
#define FLAT_STATE_END_TEMP     (1683)
#define END_COMPARE_Y_X         (1484)
#define SKIP_SWING_END_1        (1551)
#define SKIP_SWING_END_3        (1588)
#define SKIP_SWING_END_2        (1566)
#define TILTG75_START           (1672)
#define CFG_6                   (2753)
#define TILTL75_END             (1669)
#define END_ORIENT              (1884)
#define CFG_FLICK_IN            (2573)
#define TILTL75_START           (1643)
#define CFG_MOTION_BIAS         (1208)
#define X_GRT_Y                 (1408)
#define TEMPLABEL               (2324)
#define CFG_ANDROID_ORIENT_INT  (1853)
#define CFG_GYRO_RAW_DATA       (2722)
#define X_GRT_Y_TMP2            (1379)

#define D_0_22                  (22+512)
#define D_0_24                  (24+512)

#define D_0_36                  (36)
#define D_0_52                  (52)
#define D_0_96                  (96)
#define D_0_104                 (104)
#define D_0_108                 (108)
#define D_0_163                 (163)
#define D_0_188                 (188)
#define D_0_192                 (192)
[... 583 lines cut ...]
  VectorFloat rotate(Quaternion *q) {
    Quaternion p(0, x, y, z);

    // quaternion multiplication: q * p, stored back in p
    p = q -> getProduct(p);

    // quaternion multiplication: p * conj(q), stored back in p
    p = p.getProduct(q -> getConjugate());

    // p quaternion is now [0, x', y', z']
  //  x = p.x;
  //  y = p.y;
  //  z = p.z;
    //return this;
  return VectorFloat( p.x,p.y,p.z);
  }

  VectorFloat getRotated(Quaternion *q) {
    VectorFloat r(x, y, z);
    r.rotate(q);
    return r;
  }
};

#endif

Andres-Araujo, 2021-09-18:
Hi, I still have errors with the compilation:

In file included from C:\Users\PC\Documents\Arduino\sketch_sep18c\sketch_sep18c.ino:3:0:
C:\Users\PC\Documents\Arduino\libraries\Simple_MPU6050-master/MPU_WriteMacros.h:313:0: warning: "EXT_I2C_ST1_READ_BYTE" redefined
 #define EXT_I2C_ST1_READ_BYTE(Data)      MPUi2cReadByte(0x4D, (uint8_t *)Data)
 
C:\Users\PC\Documents\Arduino\libraries\Simple_MPU6050-master/MPU_WriteMacros.h:309:0: note: this is the location of the previous definition
 #define EXT_I2C_ST1_READ_BYTE(Data)      MPUi2cReadByte(0x49, (uint8_t *)Data)
 
C:\Users\PC\Documents\Arduino\libraries\Simple_MPU6050-master/MPU_WriteMacros.h:413:0: warning: "AKM_POWER_DOWN" redefined
 #define AKM_POWER_DOWN          (0x00)
 
C:\Users\PC\Documents\Arduino\libraries\Simple_MPU6050-master/MPU_WriteMacros.h:363:0: note: this is the location of the previous definition
 #define AKM_POWER_DOWN(HIGHS_SENS)           (0b0000|(HIGHS_SENS & 1)<<4)
 
C:\Users\PC\Documents\Arduino\libraries\Simple_MPU6050-master/MPU_WriteMacros.h:415:0: warning: "AKM_FUSE_ROM_ACCESS" redefined
 #define AKM_FUSE_ROM_ACCESS     (0x0F)
 
C:\Users\PC\Documents\Arduino\libraries\Simple_MPU6050-master/MPU_WriteMacros.h:369:0: note: this is the location of the previous definition
 #define AKM_FUSE_ROM_ACCESS(HIGHS_SENS)          (0b1111|(HIGHS_SENS & 1)<<4)
 
In file included from C:\Users\PC\Documents\Arduino\sketch_sep18c\sketch_sep18c.ino:1:0:
C:\Users\PC\Documents\Arduino\libraries\Simple_MPU6050-master/DMP_Image.h:575:28: warning: ISO C++ forbids declaration of 'normalize' with no type [-fpermissive]
   VectorInt16 :: normalize() {
                            ^
C:\Users\PC\Documents\Arduino\libraries\Simple_MPU6050-master/DMP_Image.h:575:3: warning: extra qualification 'VectorInt16::' on member 'normalize' [-fpermissive]
   VectorInt16 :: normalize() {
   ^~~~~~~~~~~
C:\Users\PC\Documents\Arduino\libraries\Simple_MPU6050-master/DMP_Image.h:590:38: warning: ISO C++ forbids declaration of 'rotate' with no type [-fpermissive]
   VectorInt16 :: rotate(Quaternion *q) {
                                      ^
C:\Users\PC\Documents\Arduino\libraries\Simple_MPU6050-master/DMP_Image.h:590:3: warning: extra qualification 'VectorInt16::' on member 'rotate' [-fpermissive]
   VectorInt16 :: rotate(Quaternion *q) {
   ^~~~~~~~~~~
C:\Users\PC\Documents\Arduino\libraries\Simple_MPU6050-master/DMP_Image.h: In member function 'int VectorInt16::normalize()':
C:\Users\PC\Documents\Arduino\libraries\Simple_MPU6050-master/DMP_Image.h:581:44: error: cannot convert 'VectorInt16' to 'int' in return
     return VectorInt16(x /= m,y /= m,z /= m);
                                            ^
In file included from C:\Users\PC\Documents\Arduino\sketch_sep18c\sketch_sep18c.ino:1:0:
C:\Users\PC\Documents\Arduino\libraries\Simple_MPU6050-master/DMP_Image.h: In member function 'int VectorInt16::rotate(Quaternion*)':
C:\Users\PC\Documents\Arduino\libraries\Simple_MPU6050-master/DMP_Image.h:613:35: error: cannot convert 'VectorInt16' to 'int' in return
     return VectorInt16(p.x,p.y,p.z);
                                   ^
C:\Users\PC\Documents\Arduino\libraries\Simple_MPU6050-master/DMP_Image.h: At global scope:
C:\Users\PC\Documents\Arduino\libraries\Simple_MPU6050-master/DMP_Image.h:645:28: warning: ISO C++ forbids declaration of 'normalize' with no type [-fpermissive]
   VectorFloat :: normalize() {
                            ^
C:\Users\PC\Documents\Arduino\libraries\Simple_MPU6050-master/DMP_Image.h:645:3: warning: extra qualification 'VectorFloat::' on member 'normalize' [-fpermissive]
   VectorFloat :: normalize() {
   ^~~~~~~~~~~
C:\Users\PC\Documents\Arduino\libraries\Simple_MPU6050-master/DMP_Image.h:660:38: warning: ISO C++ forbids declaration of 'rotate' with no type [-fpermissive]
   VectorFloat :: rotate(Quaternion *q) {
                                      ^
C:\Users\PC\Documents\Arduino\libraries\Simple_MPU6050-master/DMP_Image.h:660:3: warning: extra qualification 'VectorFloat::' on member 'rotate' [-fpermissive]
   VectorFloat :: rotate(Quaternion *q) {
   ^~~~~~~~~~~
C:\Users\PC\Documents\Arduino\libraries\Simple_MPU6050-master/DMP_Image.h: In member function 'int VectorFloat::normalize()':
C:\Users\PC\Documents\Arduino\libraries\Simple_MPU6050-master/DMP_Image.h:651:45: error: cannot convert 'VectorFloat' to 'int' in return
     return VectorFloat( x /= m,y /= m,z /= m);
                                             ^
C:\Users\PC\Documents\Arduino\libraries\Simple_MPU6050-master/DMP_Image.h: In member function 'int VectorFloat::rotate(Quaternion*)':
C:\Users\PC\Documents\Arduino\libraries\Simple_MPU6050-master/DMP_Image.h:674:37: error: cannot convert 'VectorFloat' to 'int' in return
      return VectorFloat( p.x,p.y,p.z);
                                     ^
In file included from C:\Users\PC\Documents\Arduino\sketch_sep18c\sketch_sep18c.ino:4:0:
C:\Users\PC\Documents\Arduino\libraries\Simple_MPU6050-master/Simple_MPU6050.h: At global scope:
C:\Users\PC\Documents\Arduino\libraries\Simple_MPU6050-master/Simple_MPU6050.h:21:5: warning: 'typedef' was ignored in this declaration
     typedef struct SensorList_s {
     ^~~~~~~
C:\Users\PC\Documents\Arduino\libraries\Simple_MPU6050-master/Simple_MPU6050.h:33:5: warning: 'typedef' was ignored in this declaration
     typedef union AccelGyro_u {
     ^~~~~~~
exit status 1
Error compilando para la tarjeta Arduino Nano.

Piero15, 2021-10-02:
> Hola. Todavía encuentro algunos errores en DMP_Image.h. Ni siquiera pude compilar el archivo de ejemplo. Aquí copio los errores que he recibido:
> 
> En archivo incluido de /Users/diego/Documents/Arduino/libraries/Simple_MPU6050-master/Simple_MPU6050.h:7:0, de /Users/ diego/Documents/Arduino/libraries/ Simple_MPU6050- master/Examples/ Simple_MPU6050_Example : 3: /Users/diego/Documents/Arduino/libraries/Simple_MPU6050-master/DMP_Image.h: En la función miembro 'int VectorInt16 :: normalize ()': / Users / diego / Documents / Arduino / libraries / Simple_MPU6050-master / DMP_Image.h: 581: 44: error: no se puede convertir 'VectorInt16' a 'int' a cambio return VectorInt16 (x / = m, y / = m, z / = m); ^ En archivo incluido de /Users/diego/Documents/Arduino/libraries/Simple_MPU6050-master/Simple_MPU6050.h:7:0, de /Users/diego/Documents/Arduino/libraries/Simple_MPU6050-master/Examples/Simple_MPU6050_Example/Simple_MPU6050_Example.ino:3: / Users/ diego/ Documents/ Arduino/ libraries/ Simple_Masterh/ DU6050 miembro VectorInt16 :: rotate (Quaternion *) ': /Users/diego/Documents/Arduino/libraries/Simple_MPU6050-master/DMP_Image.h:613:35: error: no se puede convertir' VectorInt16 'a' int 'a cambio return VectorInt16 (px , py, pz); ^ /Users/diego/Documents/Arduino/libraries/Simple_MPU6050-master/DMP_Image.h: En la función miembro 'int VectorFloat :: normalize ()': / Users / diego / Documents / Arduino / libraries / Simple_MPU6050-master / DMP_Image. h: 651: 45: error: no se puede convertir 'VectorFloat' a 'int' ^ /Users/diego/Documents/Arduino/libraries/Simple_MPU6050-master/DMP_Image.h: En la función miembro 'int VectorFloat :: rotate (Quaternion *)': / Users / diego / Documents / Arduino / libraries / Simple_MPU6050-master / DMP_Image.h: 674: 37: error: no se puede convertir 'VectorFloat' a 'int' a cambio return VectorFloat (px, py, pz); ^ estado de salida 1 Error al compilar para la placa Arduino Uno.

Hola por favor me podrias ayudar a resolver ese problema, me pasale exactamente igual no encuentro forma de solucionarlo :(

closed 2021-10-04 by ZHomeSlice

ZHomeSlice, 2021-10-04:
Try the latest version of the library i just uploaded
Z