ZHomeSlice/Simple_MPU6050 #4: Problem with OverflowProtection on ESP32

intensite, 2020-09-03:
Hello,

I am experiencing compilation issues with the re-definition of the yield function.  I am wondering if someone else had this problem and maybe a solution.  I am using a ESP32 and my buid environment is PlatformIO.

The problem seems to be related to this line in Simple_MPU6050.h
`#define ENABLE_MPU_OVERFLOW_PROTECTION(...) void yield(void){mpu.OverflowProtection();} // yield is called from within the delay() function 
`

As noted in another issue, I had to put `build_flags = -fpermissive` in the platformio.ini file.
Trying to compile the Simple_MPU6050_Example, everything compile but get the following error message at linking:

`Building in release mode
Linking .pio\build\esp32dev\firmware.elf
.pio\build\esp32dev\libFrameworkArduino.a(esp32-hal-misc.c.o): In function `yield':
C:\users\stephen\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-misc.c:48: multiple definition of `yield'
.pio\build\esp32dev\src\main.cpp.o:D:\projects\arduino\ESP32\simple_mpu6050\simple_MPU6050/src/main.cpp:10: first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1`

Note:  The Simple_MPU6050_Example.ino was renamed to main.cpp to make it easier for platformIO

ZHomeSlice, 2021-10-04:
The overflow protection items are now removed as they were replaced with a more proactive solution.

closed 2021-10-04 by ZHomeSlice

intensite, 2021-10-05:
You are on fire since the past few days, closing many issues.  :+1: