ZHomeSlice/Simple_MPU6050 #10: Mystery DMP? issue

Hasi123, 2021-02-27:
I have been testing and playing around with your code and noticed a strange behavior, where the Z world acceleration is incorrectly calculated. When laying flat it works fine, but if rotated or upside down I get values of up to +-700 even when not moving.

Here I placed the sensor in different rotation angles on the desk each for a few seconds. The problem shows up quite nicely in the values: [output.txt](https://github.com/ZHomeSlice/Simple_MPU6050/files/6054467/output.txt)

I have this problem on multiple sensors and tried recalibrating several times. I cant figure out where the problem is, since the raw sensor values seem ok.

Hasi123, 2021-03-01:
I tested it with completely different hardware. A 32U4 and a 328p as the MCU and various different MPU6050s with different ages and sources. I can reproduce the problem on every test, so we can confidently eliminate hardware failure.

Here is the output from a different board with different MCU and IMU: 
[output2.txt](https://github.com/ZHomeSlice/Simple_MPU6050/files/6063745/output2.txt)

Does no one else have this issue?

ZHomeSlice, 2021-03-01:
I wish I was an expert on the math side. the Quaternions returned from the MPU should be accurate as they are generated from the eDMP firmware provided by Invensense which can't be modified to any extent outside of their internal programming department. What is unknown to me is how the math works with the quaternion output and how to change it properly. the math is located starting at line 650 and ending at line 1000 under the heading of  Helper Math Functions. The math was copied from jeffs MPU6050 libraries. correcting this here will also correct help correct Jeff's code. My goal was to provide you with complete access to the mpu6050 attributes and simplify the setup of the eDMP firmware load. in reality, my code and jeffs code do the same thing.

If you or anyone can assist with the math I would greatly appreciate it.  I have tried to get my mind wrapped around quaternion values but this just didn't quite fit lol.

I open this up to all who would like to help.

Thanks in advance.
Z

Hasi123, 2021-03-02:
Ok after a lot of testing and experimenting I came to the conclusion, that the problem is not with the math functions or the DMP firmware, but the calibration of the accelerometer. In your code you only calibrate the accelerometer offsets, but not the scaling factors. I am referring to the [SetXAccelOffset ](https://github.com/jrowberg/i2cdevlib/blob/10bcf2b49f82feec316871ac360a7e0be25b45f8/Arduino/MPU6050/MPU6050.cpp#L2863)function in the i2cdevlib MPU6050 library, or if you like these registers:
```
#define MPU6050_RA_X_FINE_GAIN      0x03 //[7:0] X_FINE_GAIN
#define MPU6050_RA_Y_FINE_GAIN      0x04 //[7:0] Y_FINE_GAIN
#define MPU6050_RA_Z_FINE_GAIN      0x05 //[7:0] Z_FINE_GAIN
```
You have to calibrate these somehow (not easy), because the factory values are way off like the offsets.

ZHomeSlice, 2021-03-02:
Very interesting.
Do you have any documentation describing the registers? I have not discovered descriptions of these till now. Thank you.
They can esasily be added

Here are example macros for the registers (Naming could change to match the Register map document if one can be found.)
//6050 Fine Gain Offsets (Need Documentation)
#define RA_X_FINE_GAIN(Data)         MPUi2cWriteByte(0x03, (uint8_t)Data)			//[7:0] X_FINE_GAIN
#define RA_Y_FINE_GAIN(Data)         MPUi2cWriteByte(0x04, (uint8_t)Data)			//[7:0] Y_FINE_GAIN
#define RA_Z_FINE_GAIN(Data)         MPUi2cWriteByte(0x05, (uint8_t)Data)			//[7:0] Z_FINE_GAIN

These would be added at line 10 in MPU_WriteMacros.h

Now how are they used?
What do they modify and how does it affect the output. I would like to implement these into my class to further calibrate the MPU6050 accelerometer.

Z

Hasi123, 2021-03-02:
Have not found any documentation about those registers from InvenSense.

When using calibrated values from sensors it usually looks something like this:
real value = (measured value - offset) * scaling factor

These registers somehow control the scaling factor, but not 100% how they should be calculated. What I did was to measure the acceleration in every direction and came up with [this spreadsheet](https://docs.google.com/spreadsheets/d/1mSyhNnmM4mHULDqibQgkb6YQP4wc7KQ_pI2Ulx6xS-I/edit?usp=sharing) to determine the values.

ZHomeSlice, 2021-03-02:
I've added the macros to the master for both reading and writing and I updated my last comment.
Now the question is how to use them??? 
I'm going to look around in Invensense code to see if they play with those values in any way.
Z

closed 2021-10-12 by ZHomeSlice