ZHomeSlice/Simple_MPU6050 #11: How-to change the MPU6050 sample rate?

kevintedder, 2021-04-10:
The current sample rate of 100Hz is too high for what I need. Can it be changed? If so, what do I need to amend in the code to bring it down to 40Hz?

kevintedder, 2021-05-24:
[SOLVED}  I found the macro SMPLRT_DIV_WRITE_SMPLRT_DIV() in the MPU_ReadMacros.h file.   However, the DMP output appears to become more sluggish to any movement as the sample rate is reduced.   So I abandoned the idea and left it at the default 200Hz.

FYI - I found these values through trial and error using a Oscilloscope. 
 /*
        Delay   IMU Sample
        Value   Frequency    Calc:  1000/(1 + #)
        #1    = 250Hz
        #2    = 200Hz
        #8    = 111Hz
        #9    = 100Hz
        #10   = 90Hz
        #11   = 83Hz
        #12   = 76Hz
        #13   = 71Hz
        #14   = 66Hz
        #15   = 62Hz
        #16   = 58Hz
        #17   = 55Hz
        #18   = 52Hz
        #19   = 50Hz
        #20   = 48Hz
        #21   = 45Hz
        #22   = 43Hz
        #23   = 41Hz
        #24   = 40Hz - This was my objective but the IMU took 5-10 secs to settle - no good for balancing
        #25   = 38Hz
  */

closed 2021-05-24 by kevintedder

ZHomeSlice, 2021-10-04:
Changing the DMP Sample Rate

How does InvenSense achieve this

In fine inv_mpu_dmp_motion_driver.c 
from https://invensense.tdk.com/ 

```
/**
 *  @brief      Set DMP output rate.
 *  Only used when DMP is on.
 *  @param[in]  rate    Desired fifo rate (Hz).
 *  @return     0 if successful.
 */
int dmp_set_fifo_rate(unsigned short rate)
{
    const unsigned char regs_end[12] = {DINAFE, DINAF2, DINAAB, 0xc4, DINAAA, DINAF1, DINADF, DINADF, 0xBB, 0xAF, DINADF, DINADF};

/* Simplified 
    const unsigned char regs_end[12] = {0xFE, 0xF2, 0xAB, 0xc4, 0xAA, 0xF1, 0xDF, 0xDF, 0xBB, 0xAF, 0xDF, 0xDF};
*/
    unsigned short div; // uint16_t
    unsigned char tmp[8];  //uint9_t array

    if (rate > DMP_SAMPLE_RATE)
        return -1;
// #define DMP_SAMPLE_RATE     (200)   // this is the rate that the dmp gets data from the gyro and the accellerometer
    div = DMP_SAMPLE_RATE / rate - 1;  
// 200Hz = 200 / 200 -1 = 0x00 
// 100Hz = 200 / 100 -1 = 0x01 
// 50Hz = 200 / 50 -1 = 0x03 
// 40Hz = 200 / 40 -1 = 0x04 
// 25Hz = 0x07 
// 20Hz = 0x09
// 10Hz = 0x13
// 1Hz = 0xC7 
// every 10 seconds = 0x07 0xCF
// every 60 seconds = 0x2E 0xDF

    tmp[0] = (unsigned char)((div >> 8) & 0xFF);
    tmp[1] = (unsigned char)(div & 0xFF);
    if (mpu_write_mem(D_0_22, 2, tmp))  // We need to change 2 bytes in the embedded DMP
        return -1;
    if (mpu_write_mem(CFG_6, 12, (unsigned char*)regs_end))  // this is already done when I created this instance of DMP so we dont need to do it again.
        return -1;

    dmp.fifo_rate = rate;// Store the value for there program
    return 0;
}
```
what is happening:
mpu_write_mem modifies the embedded DMP image.

So let's make some changes :)

200Hz = 200 / 200 -1 = 0x00 
100Hz = 200 / 100 -1 = 0x01 
50Hz = 200 / 50 -1 = 0x03
40Hz = 200 / 40 -1 = 0x04 
25Hz = 0x07 
20Hz = 0x09
10Hz = 0x13
1Hz = 0xC7 
every 10 seconds = 0x07 0xCF
every 60 seconds = 0x2E 0xDF

Memory Location in DMP BLOB
#define D_0_22                  (22+512)  // Already set for us
I have a cool function already set for us to do this

mpu.write_mem(uint16_t  mem_addr, uint16_t  length, uint8_t *Data)

uint9_t DMP_Output_Rate[{0x00,0x01];  // 100 HZ Default

#define DMP_200Hz  DMP_Output_Rate[0] = 0x00; DMP_Output_Rate[1] = 0x00;
#define DMP_100Hz  DMP_Output_Rate[0] = 0x00; DMP_Output_Rate[1] = 0x01;
#define DMP_50Hz DMP_Output_Rate[0] = 0x00; DMP_Output_Rate[1] = 0x03;
#define DMP_40Hz DMP_Output_Rate[0] = 0x00; DMP_Output_Rate[1] = 0x04;
#define DMP_25Hz DMP_Output_Rate[0] = 0x00; DMP_Output_Rate[1] = 0x07;
#define DMP_20Hz DMP_Output_Rate[0] = 0x00; DMP_Output_Rate[1] = 0x09;
#define DMP_10Hz DMP_Output_Rate[0] = 0x00; DMP_Output_Rate[1] = 0x13;
#define DMP_1Hz DMP_Output_Rate[0] = 0x00; DMP_Output_Rate[1] = 0xC7;
#define DMP_1sec DMP_Output_Rate[0] = 0x00; DMP_Output_Rate[1] = 0xC7;
#define DMP_10sec  DMP_Output_Rate[1] = 0x07; DMP_Output_Rate[1] = 0xCF;
#define DMP_60sec  DMP_Output_Rate[1] = 0x2E; DMP_Output_Rate[1] = 0xDF;

mpu.write_mem(D_0_22, 2, DMP_Output_Rate);

ZHomeSlice, 2021-10-04:
@kevintedder 
You will like this 

Take a look at the updated code 
Just added
`mpu.Set_DMP_Output_Rate(DMP_100Hz);// Set the DMP output rate from 200Hz to 5 Minutes`

Here are the pre-calculated  values:
```
#define DMP_200Hz  0x00, 0x00
#define DMP_100Hz  0x00, 0x01
#define DMP_50Hz   0x00, 0x03
#define DMP_40Hz   0x00, 0x04
#define DMP_25Hz   0x00, 0x07
#define DMP_20Hz   0x00, 0x09
#define DMP_10Hz   0x00, 0x13
#define DMP_1Hz    0x00, 0xC7
#define DMP_1Sec   0x00, 0xC7
#define DMP_10Sec  0x07, 0xCF
#define DMP_60Sec  0x2E, 0xDF
#define DMP_1Min   0x2E, 0xDF
#define DMP_5Min   0xEA, 0x5F 
```
Enjoy
Z

ZHomeSlice, 2021-10-11:
@kevintedder 
Just added a function to adjust DMP output rate from 200Hz to 5 minutes.
look for the following options for ease of use.
```
  mpu.Set_DMP_Output_Rate_Hz(10);           // Set the DMP output rate from 200Hz to 5 Minutes.
  mpu.Set_DMP_Output_Rate_Seconds(10);   // Set the DMP output rate in Seconds
  mpu.Set_DMP_Output_Rate_Minutes(5);    // Set the DMP output rate in Minutes
```