ZHomeSlice/Simple_MPU6050 #13: Simple_MPU6050/DMP_Image.h:575:3: error: ISO C++ forbids declaration

fanlessfan, 2021-09-29:
Hello,

I am trying to compile Simple_MPU6050_Example_No_interrupt_pin_used.ino (https://forum.arduino.cc/uploads/short-url/qY4D0CYmR2l0wQzHyoaMQahJVtW.ino) on Wemos D1 mini and got the error below.

Thanks in advance for the help.

`/Volumes/Shared/Documents/Arduino/libraries/Simple_MPU6050/DMP_Image.h:575:3: error: ISO C++ forbids declaration of 'normalize' with no type [-fpermissive]
  575 |   VectorInt16 :: normalize() {
      |   ^~~~~~~~~~~
/Volumes/Shared/Documents/Arduino/libraries/Simple_MPU6050/DMP_Image.h:575:3: error: extra qualification 'VectorInt16::' on member 'normalize' [-fpermissive]
/Volumes/Shared/Documents/Arduino/libraries/Simple_MPU6050/DMP_Image.h:590:3: error: ISO C++ forbids declaration of 'rotate' with no type [-fpermissive]
  590 |   VectorInt16 :: rotate(Quaternion *q) {
      |   ^~~~~~~~~~~
/Volumes/Shared/Documents/Arduino/libraries/Simple_MPU6050/DMP_Image.h:590:3: error: extra qualification 'VectorInt16::' on member 'rotate' [-fpermissive]
/Volumes/Shared/Documents/Arduino/libraries/Simple_MPU6050/DMP_Image.h: In member function 'int VectorInt16::normalize()':
/Volumes/Shared/Documents/Arduino/libraries/Simple_MPU6050/DMP_Image.h:581:12: error: cannot convert 'VectorInt16' to 'int' in return
  581 |     return VectorInt16(x /= m,y /= m,z /= m);
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |            |
      |            VectorInt16
In file included from /Volumes/Shared/Documents/Arduino/libraries/Simple_MPU6050/Simple_MPU6050.h:7,
                 from /Volumes/Shared/Downloads/Simple_MPU6050_Example_No_interrupt_pin_used/Simple_MPU6050_Example_No_interrupt_pin_used.ino:3:
/Volumes/Shared/Documents/Arduino/libraries/Simple_MPU6050/DMP_Image.h: In member function 'int VectorInt16::rotate(Quaternion*)':
/Volumes/Shared/Documents/Arduino/libraries/Simple_MPU6050/DMP_Image.h:613:12: error: cannot convert 'VectorInt16' to 'int' in return
  613 |     return VectorInt16(p.x,p.y,p.z);
      |            ^~~~~~~~~~~~~~~~~~~~~~~~
      |            |
      |            VectorInt16
/Volumes/Shared/Documents/Arduino/libraries/Simple_MPU6050/DMP_Image.h: At global scope:
/Volumes/Shared/Documents/Arduino/libraries/Simple_MPU6050/DMP_Image.h:645:3: error: ISO C++ forbids declaration of 'normalize' with no type [-fpermissive]
  645 |   VectorFloat :: normalize() {
      |   ^~~~~~~~~~~
/Volumes/Shared/Documents/Arduino/libraries/Simple_MPU6050/DMP_Image.h:645:3: error: extra qualification 'VectorFloat::' on member 'normalize' [-fpermissive]
/Volumes/Shared/Documents/Arduino/libraries/Simple_MPU6050/DMP_Image.h:660:3: error: ISO C++ forbids declaration of 'rotate' with no type [-fpermissive]
  660 |   VectorFloat :: rotate(Quaternion *q) {
      |   ^~~~~~~~~~~
/Volumes/Shared/Documents/Arduino/libraries/Simple_MPU6050/DMP_Image.h:660:3: error: extra qualification 'VectorFloat::' on member 'rotate' [-fpermissive]
/Volumes/Shared/Documents/Arduino/libraries/Simple_MPU6050/DMP_Image.h: In member function 'int VectorFloat::normalize()':
/Volumes/Shared/Documents/Arduino/libraries/Simple_MPU6050/DMP_Image.h:651:12: error: cannot convert 'VectorFloat' to 'int' in return
  651 |     return VectorFloat( x /= m,y /= m,z /= m);
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |            |
      |            VectorFloat
/Volumes/Shared/Documents/Arduino/libraries/Simple_MPU6050/DMP_Image.h: In member function 'int VectorFloat::rotate(Quaternion*)':
/Volumes/Shared/Documents/Arduino/libraries/Simple_MPU6050/DMP_Image.h:674:13: error: cannot convert 'VectorFloat' to 'int' in return
  674 |      return VectorFloat( p.x,p.y,p.z);
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~
      |             |
      |             VectorFloat
exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.`

ZHomeSlice, 2021-09-29:
The errors are caused by my inexperience in programming. The -fpermissive  flag will downgrade some diagnostics about nonconformant code from errors to warnings. Thus, using -fpermissive will allow some nonconforming code to compile.

For those with better programming knowledge, helping me correct this nonconformant code would be great. 

hope this helps
Z

fanlessfan, 2021-09-30:
Thanks Z. Could you let me know how to use -fpermissive on Arduino IDE?

ZHomeSlice, 2021-10-03:
Try it out now

fanlessfan, 2021-10-04:
Thank you so much @ZHomeSlice  the compiling passed. how error I still got error when running the example with both MPU6050 and MPU9250.

User exception (panic/abort/assert)
1:02:59.445 -> User exception (panic/abort/assert)
11:02:59.445 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
11:02:59.478 -> 
11:02:59.478 -> Abort called
11:02:59.478 -> 
11:02:59.478 -> >>>stack>>>
11:02:59.478 -> 
11:02:59.478 -> ctx: cont
11:02:59.478 -> sp: 3ffffef0 end: 3fffffc0 offset: 0000
11:02:59.478 -> 3ffffef0:  00000001 00000068 00000068 40202f28  
11:02:59.478 -> 3fffff00:  000000fe 00000000 00000000 00000000  
11:02:59.478 -> 3fffff10:  00000000 00000000 00000000 00ff0000  
11:02:59.478 -> 3fffff20:  5ffffe00 5ffffe00 3ffee944 3ffee834  
11:02:59.478 -> 3fffff30:  00000000 00000001 00000002 4020468a  
11:02:59.513 -> 3fffff40:  40100679 3ffee944 0000000a 4020469c  
11:02:59.513 -> 3fffff50:  40204ead 0000000a 00000002 40205295  
11:02:59.513 -> 3fffff60:  3fffdad0 00000000 3ffee58c 40201784  
11:02:59.513 -> 3fffff70:  3fffdad0 00000000 3ffee58c 40205334  
11:02:59.513 -> 3fffff80:  3fffdad0 00000000 3ffee58c 40202c80  
11:02:59.513 -> 3fffff90:  3fffdad0 00000000 3ffee58c 40201241  
11:02:59.513 -> 3fffffa0:  feefeffe feefeffe 3ffee820 40203fb0  
11:02:59.513 -> 3fffffb0:  feefeffe feefeffe 3ffe85e4 40100f35  
11:02:59.548 -> <<<stack<<<
11:02:59.548 -> 
11:02:59.548 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
11:02:59.548 ->

fanlessfan, 2021-10-04:
Thank you so much @ZHomeSlice . now there are new errors below.

Simple_MPU6050_Example_No_interrupt_pin_used:9:33: error: expected constructor, destructor, or type conversion before ';' token

====full error=====
Arduino: 1.8.16 (Mac OS X), Board: "Generic ESP8266 Module, 80 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, dtr (aka nodemcu), 26 MHz, 40MHz, DOUT (compatible), 1MB (FS:64KB OTA:~470KB), 2, nonos-sdk 2.2.1+100 (190703), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

/Volumes/data/apps/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Volumes/data/apps/Arduino.app/Contents/Java/hardware -hardware /Users/fzhou/Library/Arduino15/packages -tools /Volumes/data/apps/Arduino.app/Contents/Java/tools-builder -tools /Volumes/data/apps/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/fzhou/Library/Arduino15/packages -built-in-libraries /Volumes/data/apps/Arduino.app/Contents/Java/libraries -libraries /Users/fzhou/Documents/Arduino/libraries -fqbn=esp8266:esp8266:generic:xtal=80,vt=flash,exception=disabled,stacksmash=disabled,ssl=all,mmu=3232,non32xfer=fast,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=1M64,led=2,sdk=nonosdk_190703,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -vid-pid=0000_0000 -ide-version=10816 -build-path /var/folders/wg/hrp4gsks3n1bt34wc0w3pl680000gp/T/arduino_build_216013 -warnings=none -build-cache /var/folders/wg/hrp4gsks3n1bt34wc0w3pl680000gp/T/arduino_cache_587836 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=/Users/fzhou/Library/Arduino15/packages/esp8266/tools/mkspiffs/3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mkspiffs-3.0.4-gcc10.3-1757bed.path=/Users/fzhou/Library/Arduino15/packages/esp8266/tools/mkspiffs/3.0.4-gcc10.3-1757bed -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/Users/fzhou/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed -prefs=runtime.tools.xtensa-lx106-elf-gcc-3.0.4-gcc10.3-1757bed.path=/Users/fzhou/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed -prefs=runtime.tools.python3.path=/Users/fzhou/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=/Users/fzhou/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1 -prefs=runtime.tools.mklittlefs.path=/Users/fzhou/Library/Arduino15/packages/esp8266/tools/mklittlefs/3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mklittlefs-3.0.4-gcc10.3-1757bed.path=/Users/fzhou/Library/Arduino15/packages/esp8266/tools/mklittlefs/3.0.4-gcc10.3-1757bed -verbose /Volumes/personal/fzhou/Arduino/Simple_MPU6050_Example_No_interrupt_pin_used/Simple_MPU6050_Example_No_interrupt_pin_used.ino
/Volumes/data/apps/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Volumes/data/apps/Arduino.app/Contents/Java/hardware -hardware /Users/fzhou/Library/Arduino15/packages -tools /Volumes/data/apps/Arduino.app/Contents/Java/tools-builder -tools /Volumes/data/apps/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/fzhou/Library/Arduino15/packages -built-in-libraries /Volumes/data/apps/Arduino.app/Contents/Java/libraries -libraries /Users/fzhou/Documents/Arduino/libraries -fqbn=esp8266:esp8266:generic:xtal=80,vt=flash,exception=disabled,stacksmash=disabled,ssl=all,mmu=3232,non32xfer=fast,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=1M64,led=2,sdk=nonosdk_190703,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -vid-pid=0000_0000 -ide-version=10816 -build-path /var/folders/wg/hrp4gsks3n1bt34wc0w3pl680000gp/T/arduino_build_216013 -warnings=none -build-cache /var/folders/wg/hrp4gsks3n1bt34wc0w3pl680000gp/T/arduino_cache_587836 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=/Users/fzhou/Library/Arduino15/packages/esp8266/tools/mkspiffs/3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mkspiffs-3.0.4-gcc10.3-1757bed.path=/Users/fzhou/Library/Arduino15/packages/esp8266/tools/mkspiffs/3.0.4-gcc10.3-1757bed -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/Users/fzhou/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed -prefs=runtime.tools.xtensa-lx106-elf-gcc-3.0.4-gcc10.3-1757bed.path=/Users/fzhou/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed -prefs=runtime.tools.python3.path=/Users/fzhou/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=/Users/fzhou/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1 -prefs=runtime.tools.mklittlefs.path=/Users/fzhou/Library/Arduino15/packages/esp8266/tools/mklittlefs/3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mklittlefs-3.0.4-gcc10.3-1757bed.path=/Users/fzhou/Library/Arduino15/packages/esp8266/tools/mklittlefs/3.0.4-gcc10.3-1757bed -verbose /Volumes/personal/fzhou/Arduino/Simple_MPU6050_Example_No_interrupt_pin_used/Simple_MPU6050_Example_No_interrupt_pin_used.ino
Using board 'generic' from platform in folder: /Users/fzhou/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2
Using core 'esp8266' from platform in folder: /Users/fzhou/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2
Detecting libraries used...
[... cut ...]
  110 |       mpu.OverflowProtection(); // add this line to keep the FIFO buffer from overflowing while we wait
      |           ^~~~~~~~~~~~~~~~~~
Using library Simple_MPU6050 in folder: /Users/fzhou/Documents/Arduino/libraries/Simple_MPU6050 (legacy)
Using library Wire at version 1.0 in folder: /Users/fzhou/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/Wire 
Using library I2Cdev in folder: /Users/fzhou/Documents/Arduino/libraries/I2Cdev (legacy)
exit status 1
expected constructor, destructor, or type conversion before ';' token

ZHomeSlice, 2021-10-05:
So I have a ESP8266 and I have many MPU6050's but I have never tried to use them together :)
I will need to see how you are attaching them electrically. Please prived a schematic.
I will also need to test code 
What GUI are you using to get this to work?
Please provide an example code to test.
Z

note that 'OverflowProtection' was completely removed. you will need to update your scripts.

fanlessfan, 2021-10-05:
Thank you @ZHomeSlice 

I use D1-mini with GY-521 MPU6050 board without interrupt pin. using an example code from Simple_MPU6050_Example_No_interrupt_pin_used.ino (https://forum.arduino.cc/uploads/short-url/qY4D0CYmR2l0wQzHyoaMQahJVtW.ino). I use Arduino IDE 1.8.16 (https://www.arduino.cc/en/software). I wonder if you can provide no interrupt pin example.

D1-MINI   MPU6050
5V------> VCC
G ------> GND
SCL----> D4 (GPIO2)
SDA----> D3 (GPIO0)

ZHomeSlice, 2021-10-05:
This may take a few days before I can build the unit I'm ordering the D1-mini. I like the size of that development board.

Z

fanlessfan, 2021-10-05:
Thanks @ZHomeSlice I think you existing ESP8266 should be the same. I just need a no interrupt pin example to use your library.

ZHomeSlice, 2021-10-05:
@fanlessfan 
so I just updated the entire code base
Nothing I did should affect you adversely

to prevent the pin from being added
in Simple_MPU6050.h
on line 4 Remark out 
#define interruptPin 2

this will remove all interrupt settings.

try this example:
https://github.com/ZHomeSlice/Simple_MPU6050/blob/master/Examples/Simple_MPU6050_Basic_Example/Simple_MPU6050_Basic_Example.ino

it doesn't use interrupts and is as simple as possible.
Z

fanlessfan, 2021-10-05:
Thank you @ZHomeSlice for your quick response. 

I compiled the basic_Example and got error below. I also remove the interruptPin
  class TwoWire' has no member named 'setWireTimeout'; did you mean 'setTimeout'?

If I remove the code like below. I passed the compile and got run time error

void setup() {
  uint8_t val;
  // join I2C bus (I2Cdev library doesn't do this automatically)
#if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
  Wire.begin(0,2);
  Wire.setClock(400000); // 400kHz I2C clock. Comment this line if having compilation difficulties
  //Wire.setWireTimeout(3000, true); //timeout value in uSec <---------------comment out this line

run time error:
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (0):
epc1=0x402071a2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffe00 end: 3fffffc0 offset: 0190
3fffff90:  3fffdad0 00000000 3ffee7c8 4020122c  
3fffffa0:  feefeffe feefeffe 3ffee81c 40203ff8  
3fffffb0:  feefeffe feefeffe 3ffe85e4 40100f35  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v00046970
~ld
Start:

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (0):
epc1=0x402071a2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffe00 end: 3fffffc0 offset: 0190
3fffff90:  3fffdad0 00000000 3ffee7c8 4020122c  
3fffffa0:  feefeffe feefeffe 3ffee81c 40203ff8  
3fffffb0:  feefeffe feefeffe 3ffe85e4 40100f35  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

ZHomeSlice, 2021-10-06:
Remove it
Just added it to fix issues related to:

"the classic I2C bus 'freeze' problem, and if so, it has nothing at all to do with the MPU6060. I2C bus freezes are a known problem with Arduino, and it has only been recently addressed. See this post for all the gory details. The good news is, if you use the latest Wire library code and add the timeout setting line to your setup() function like this

Wire.setWireTimeout(3000, true); //timeout value in uSec

then the I2C freeze problem should go away. Note that if you aren't using the latest Wire library, the above code line won't compile, so it's easy to tell one way or the other."
[https://github.com/jrowberg/i2cdevlib/issues/543#issuecomment-707336866](https://github.com/jrowberg/i2cdevlib/issues/543#issuecomment-707336866)

ZHomeSlice, 2021-10-10:
@fanlessfan 
Got it to work :)
I'll Send the temporary fix shortly
Hoping to implement a new example into my library
Z

ZHomeSlice, 2021-10-10:
Ok Here you go
You can remove the wifi setup stuff and only watch the serial port. I kept the other as it worked and I didn't want to degrade the usefulness of the WiFi connectivity of the esp8266!

in Simple_MPU6050.h
remark out line 4
`//#define interruptPin 2`

This is the sketch I am using it is an adjustment to jeffs MPU6050's library example sketch MPU6050_DMP6_ESPWiFi.ino 
```
/* ============================================
  I2Cdev device library code is placed under the MIT license
  Copyright (c) 2012 Jeff Rowberg

  Permission is hereby granted, free of charge, to any person obtaining a copy
  of this software and associated documentation files (the "Software"), to deal
  in the Software without restriction, including without limitation the rights
  to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
  copies of the Software, and to permit persons to whom the Software is
  furnished to do so, subject to the following conditions:

  The above copyright notice and this permission notice shall be included in
  all copies or substantial portions of the Software.

  THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
  IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
  FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
  AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
  LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
  OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
  THE SOFTWARE.
  ===============================================
*/

/* This driver reads quaternion data from the MPU6060 and sends
   Open Sound Control messages.

  GY-521  NodeMCU
  MPU6050 devkit 1.0
  board   Lolin         Description
  ======= ==========    ====================================================
  VCC     VU (5V USB)   Not available on all boards so use 3.3V if needed.
  GND     G             Ground
  SCL     D1 (GPIO05)   I2C clock
  SDA     D2 (GPIO04)   I2C data
  XDA     not connected
  XCL     not connected
  AD0     not connected
  INT     D8 (GPIO15)   Interrupt pin

*/

#if defined(ESP8266)
#include <ESP8266WiFi.h>
#else
#include <WiFi.h>
#endif
#include <DNSServer.h>
#include <WiFiClient.h>
#include <WiFiUdp.h>
#include <OSCMessage.h>
#include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager

// I2Cdev and MPU6050 must be installed as libraries, or else the .cpp/.h files
// for both classes must be in the include path of your project
#include "I2Cdev.h"

#define MPU6050_DEFAULT_ADDRESS     0x68 // address pin low (GND), default for InvenSense evaluation board
#include "Simple_MPU6050.h"

// Arduino Wire library is required if I2Cdev I2CDEV_ARDUINO_WIRE implementation
[... 189 lines cut ...]
void loop(void)
{
  if (WiFi.status() != WL_CONNECTED) {
    Serial.println();
    Serial.println("*** Disconnected from AP so rebooting ***");
    Serial.println();
    ESP.reset();
  }
  delay(1); // required to allow the esp8266 to do other things
  mpu.dmp_read_fifo(false); // false = no interrupt pin attachment required.
  // Tests for Data in the FIFO Buffer
  // when it finds data it runs the mpu.on_FIFO(print_Values)
  // functin which we set run the print_Values Function
  // The print_Values function MUST have the following variables available to attach data
  // void print_Values (int16_t *gyro, int16_t *accel, int32_t *quat, uint32_t *timestamp)
  // Variables:
  // int16_t *gyro for the gyro values to be passed to it (The * tells the function it will be a pointer to the value)
  // int16_t *accel for the accel values to be passed to it
  // int32_t *quat for the quaternion values to be passed to it
  // uint32_t *timestamp which will be the micros()value at the time we retrieved the Newest value from the FIFO Buffer.
}
```
ToDo: I will work on removing the `#define interruptPin 2` in the .h file and make it a callback function or something to simplify setup across multiple platforms. 

Z