ZHomeSlice/Simple_MPU6050 #19: How can i config the FULL SCALE SELECT of MPU6050 module?

JeffJiang1024, 2021-11-30:
Hello!
I'm glad to know your project these days and it is helpful to me.
I noticed that you have done some define in "MPU_ReadMacros.h" like this: 
``` c
#define GYRO_CONFIG_READ_XGYRO_Ct_en(Data)					MPUi2cRead(0x1B, 1, 7, (uint8_t *)Data)  //   X Gyro self-test
#define GYRO_CONFIG_READ_YGYRO_Ct_en(Data)					MPUi2cRead(0x1B, 1, 6, (uint8_t *)Data)  //   Y Gyro self-test
#define GYRO_CONFIG_READ_ZGYRO_Ct_en(Data)					MPUi2cRead(0x1B, 1, 5, (uint8_t *)Data)  //   Z Gyro self-test
#define GYRO_CONFIG_READ_GYRO_FS_SEL(Data)					MPUi2cRead(0x1B, 2, 4, (uint8_t *)Data)  //   Gyro Full Scale Select: 00B = +250 01B = +500 10B = +1000 11B = 2000

#define GYRO_CONFIG_READ_FCHOICE_B(Data)					MPUi2cRead(0x1B, 2, 2, (uint8_t *)Data)  //   Used to bypass DLPF
#define ACCEL_CONFIG_READ_ax_st_en(Data)					MPUi2cRead(0x1C, 1, 7, (uint8_t *)Data)  //   X Accel self-test
#define ACCEL_CONFIG_READ_ay_st_en(Data)					MPUi2cRead(0x1C, 1, 6, (uint8_t *)Data)  //   Y Accel self-test
#define ACCEL_CONFIG_READ_az_st_en(Data)					MPUi2cRead(0x1C, 1, 5, (uint8_t *)Data)  //   Z Accel self-test
#define ACCEL_CONFIG_READ_ACCEL_FS_SEL(Data)				MPUi2cRead(0x1C, 2, 3, (uint8_t *)Data)  //   Accel Full Scale Select:  2g (00),  4g (01),  8g (10),  16g (11)
```
But I couldn't find where and how can I config GYRO and ACCEL Full-Scale Range, wish you can help me!
Thanks a lot!

ZHomeSlice, 2021-12-16:
How to use my macros
first, get the registry map document
https://invensense.tdk.com/wp-content/uploads/2015/02/MPU-6000-Register-Map1.pdf

Search for what you are looking for:
I looked for "Full Scale"
it brought me to 
Register 27 – Gyroscope Configuration
GYRO_CONFIG

In the macros for Writing to the MPU I searched for GYRO_CONFIG and got:

```
#define GYRO_CONFIG_WRITE_XGYRO_Ct_en(Data)             MPUi2cWrite(0x1B, 1, 7, (uint8_t)Data)		//   X Gyro self-test
#define GYRO_CONFIG_WRITE_YGYRO_Ct_en(Data)             MPUi2cWrite(0x1B, 1, 6, (uint8_t)Data)		//   Y Gyro self-test
#define GYRO_CONFIG_WRITE_ZGYRO_Ct_en(Data)             MPUi2cWrite(0x1B, 1, 5, (uint8_t)Data)		//   Z Gyro self-test
#define GYRO_CONFIG_WRITE_GYRO_FS_SEL(Data)             MPUi2cWrite(0x1B, 2, 4, (uint8_t)Data)		//   Gyro Full Scale Select: 00B = +250 01B = +500 10B = +1000 11B = 2000

#define GYRO_CONFIG_WRITE_GYRO_FS_SEL_250(...)          MPUi2cWrite(0x1B, 3, 4, (uint8_t)0)			//   Gyro Full Scale Select: +250 Deg/sec
#define GYRO_CONFIG_WRITE_GYRO_FS_SEL_500(...)          MPUi2cWrite(0x1B, 3, 4, (uint8_t)1)			//   Gyro Full Scale Select: +500 Deg/sec
#define GYRO_CONFIG_WRITE_GYRO_FS_SEL_1000(...)         MPUi2cWrite(0x1B, 3, 4, (uint8_t)2)			//   Gyro Full Scale Select: +1000 Deg/sec
#define GYRO_CONFIG_WRITE_GYRO_FS_SEL_2000(...)         MPUi2cWrite(0x1B, 3, 4, (uint8_t)3)			//   Gyro Full Scale Select: +2000 Deg/sec
#define GYRO_CONFIG_WRITE_FCHOICE_B(Data)               MPUi2cWrite(0x1B, 2, 2, (uint8_t)Data)		//   Used to bypass DLPF
```
to set the gyro all you need to do is use this macro
`#define GYRO_CONFIG_WRITE_GYRO_FS_SEL(Data)             MPUi2cWrite(0x1B, 2, 4, (uint8_t)Data)		//   Gyro Full Scale Select: 00B = +250 01B = +500 10B = +1000 11B = 2000`

How?
`mpu.GYRO_CONFIG_WRITE_GYRO_FS_SEL_1000();`
or
`mpu.GYRO_CONFIG_WRITE_GYRO_FS_SEL(11B); // for 2000`

and for the accelerometer
Register 28 – Accelerometer Configuration
ACCEL_CONFIG

```
#define ACCEL_CONFIG_WRITE_ax_st_en(Data)               MPUi2cWrite(0x1C, 1, 7, (uint8_t)Data)		//   X Accel self-test
#define ACCEL_CONFIG_WRITE_ay_st_en(Data)               MPUi2cWrite(0x1C, 1, 6, (uint8_t)Data)		//   Y Accel self-test
#define ACCEL_CONFIG_WRITE_az_st_en(Data)               MPUi2cWrite(0x1C, 1, 5, (uint8_t)Data)		//   Z Accel self-test
#define ACCEL_CONFIG_WRITE_ACCEL_FS_SEL(Data)           MPUi2cWrite(0x1C, 2, 3, (uint8_t)Data)		//   Accel Full Scale Select: 2g (00B), 4g (01B), 8g (10B), 16g (11B)

#define ACCEL_CONFIG_WRITE_ACCEL_FS_SEL_2g(...)         MPUi2cWrite(0x1C, 2, 3, (uint8_t)0)			//   Accel Full Scale Select: 2g
#define ACCEL_CONFIG_WRITE_ACCEL_FS_SEL_4g(...)         MPUi2cWrite(0x1C, 2, 3, (uint8_t)1)			//   Accel Full Scale Select: 4g
#define ACCEL_CONFIG_WRITE_ACCEL_FS_SEL_8g(...)         MPUi2cWrite(0x1C, 2, 3, (uint8_t)2)			//   Accel Full Scale Select: 8g
#define ACCEL_CONFIG_WRITE_ACCEL_FS_SEL_16g(...)        MPUi2cWrite(0x1C, 2, 3, (uint8_t)3)			//   Accel Full Scale Select: 16g

#define ACCEL_CONFIG_2_WRITE_ACCEL_FCHOICE_B(Data)      MPUi2cWrite(0x1D, 2, 2, (uint8_t)Data)		//   Used to bypass DLPF
#define ACCEL_CONFIG_2_WRITE_A_DLPF_CFG(Data)           MPUi2cWrite(0x1D, 2, 1, (uint8_t)Data)		//   Accelerometer low pass filter setting
```
here you could use something like
`mpuACCEL_CONFIG_WRITE_ACCEL_FS_SEL_2g();`

or
`mpu.ACCEL_CONFIG_WRITE_ACCEL_FS_SEL(11B) // for 16g`

hope this helps
use the registry map to understand what you are doing and how it affects other stuff

Note: DMP requires specific values and they are set by default.
Z

JeffJiang1024, 2021-12-30:
Thank you very much for your answer, it really helped me a lot!
I have got the register map document, but I haven’t read it yet. No wonder I can’t understand this part.

Now I can continue my project happily, thanks again!

closed 2021-12-30 by JeffJiang1024