ZHomeSlice/Simple_MPU6050 #25: Compile error ?

Wbiu, 2022-08-25:
I was realy relive when found this, but some it gave errors when compiling
in Simple_Wire.cpp

26  Simple_Wire::Simple_Wire() { <----( "no default constructor exists for class "TwoWire"C/C++(291)" and "no matching function for 
27    begin();                                                                                                                                                 call to 'TwoWire::TwoWire()" ) 
28    setClock(400000); // 400kHz I2C clock. 
29    setWireTimeout(3000, true); //timeout value in uSec <----- " identifier "setWireTimeout" is undefinedC/C++(20) "
30 }
31
32 Simple_Wire::Simple_Wire(uint8_t address) { <---- " no matching function for call to 'TwoWire::TwoWire()'"
33    SetAddress(address);
34  }

I just onyl  the respo in to my MPU 6050 project. 
I´m the PlattformIO on a ESP 32 board.
Any Idea? pls

Error log : 
src/Simple_MPU6050-master/Simple_Wire.cpp: In constructor 'Simple_Wire::Simple_Wire()':
src/Simple_MPU6050-master/Simple_Wire.cpp:26:26: error: no matching function for call to 'TwoWire::TwoWire()'
 Simple_Wire::Simple_Wire() {
                          ^
In file included from src/Simple_MPU6050-master/Simple_Wire.h:28,
                 from src/Simple_MPU6050-master/Simple_Wire.cpp:24:
C:/Users/Biu/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:73:5: note: candidate: 'TwoWire::TwoWire(uint8_t)'
     TwoWire(uint8_t bus_num);
     ^~~~~~~
C:/Users/Biu/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:73:5: note:   candidate expects 1 argument, 0 provided
C:/Users/Biu/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:43:7: note: candidate: 'constexpr TwoWire::TwoWire(const TwoWire&)'
 class TwoWire: public Stream
       ^~~~~~~
C:/Users/Biu/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:43:7: note:   candidate expects 1 argument, 0 provided
src/Simple_MPU6050-master/Simple_Wire.cpp: In constructor 'Simple_Wire::Simple_Wire(uint8_t)':
src/Simple_MPU6050-master/Simple_Wire.cpp:32:41: error: no matching function for call to 'TwoWire::TwoWire()'
 Simple_Wire::Simple_Wire(uint8_t address) {
                                         ^
In file included from src/Simple_MPU6050-master/Simple_Wire.h:28,
                 from src/Simple_MPU6050-master/Simple_Wire.cpp:24:
C:/Users/Biu/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:73:5: note: candidate: 'TwoWire::TwoWire(uint8_t)'
     TwoWire(uint8_t bus_num);
     ^~~~~~~
C:/Users/Biu/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:73:5: note:   candidate expects 1 argument, 0 provided
C:/Users/Biu/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:43:7: note: candidate: 'constexpr TwoWire::TwoWire(const TwoWire&)'
 class TwoWire: public Stream
       ^~~~~~~
C:/Users/Biu/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:43:7: note:   candidate expects 1 argument, 0 provided
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\StreamString.cpp.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\USB.cpp.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\USBCDC.cpp.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\USBMSC.cpp.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\WString.cpp.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\base64.cpp.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\cbuf.cpp.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-adc.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-bt.c.o
*** [.pio\build\esp-wrover-kit\src\Simple_MPU6050-master\Simple_Wire.cpp.o] Error 1
In file included from .pio/libdeps/esp-wrover-kit/Adafruit MPU6050/Adafruit_MPU6050.cpp:38:
.pio/libdeps/esp-wrover-kit/Adafruit MPU6050/Adafruit_MPU6050.h:21:10: fatal error: Adafruit_BusIO_Register.h: No such file or directory

ZHomeSlice, 2022-08-26:
Simple_wire.cpp 
I just added this as I struggled to get jeffs i2cdev to work 100%. 
It looks like you are using it with ESP32, is this correct?

I should be able to modify or add a library to handle other boards. Currently, I'm focusing on the original Arduino UNO and Mega boards.  

Let me know what you are compiling this for.
Z

Wbiu, 2022-08-26:
Hi, 
yes Im using the ESP32. 

(Side note : I´m kinda new to the Ardruino and C++ platforom, but I´m have been coding java for a semester now.)

Was trying to the MPU6050/MPU9250 for my project, and I need a stable value from the IMU, since to raw value ( accelrometer , Gyrp ... ) are not real good , if you know what I mean.

I was searching for a solution to gat a stable value to work work with. The first solution I found was the " complementary filter"
but those libs/file dont seem to  work for me  some how.... and the secound solution  I found was the DMP solution but it was too hard to implement (-.-)  and then I fond this Lib , I was soooo happy that I found a solution but I woudnt work... I dont know what to do anymore. I just want a filter to smoothen the out put.

Any simple solution out there ?

ZHomeSlice, 2022-08-26:
I've shrunk it down to these errors:
```
C:\Users\zhome\Documents\Arduino\libraries\Simple_Wire\Simple_Wire.cpp: In constructor 'Simple_Wire::Simple_Wire()':
C:\Users\zhome\Documents\Arduino\libraries\Simple_Wire\Simple_Wire.cpp:26:26: error: no matching function for call to 'TwoWire::TwoWire()'
 Simple_Wire::Simple_Wire() {
                          ^
In file included from C:\Users\zhome\Documents\Arduino\libraries\Simple_Wire\Simple_Wire.h:28,
                 from C:\Users\zhome\Documents\Arduino\libraries\Simple_Wire\Simple_Wire.cpp:24:
C:\Users\zhome\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\Wire\src/Wire.h:73:5: note: candidate: 'TwoWire::TwoWire(uint8_t)'
     TwoWire(uint8_t bus_num);
     ^~~~~~~
C:\Users\zhome\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\Wire\src/Wire.h:73:5: note:   candidate expects 1 argument, 0 provided
C:\Users\zhome\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\Wire\src/Wire.h:43:7: note: candidate: 'constexpr TwoWire::TwoWire(const TwoWire&)'
 class TwoWire: public Stream
       ^~~~~~~
C:\Users\zhome\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\Wire\src/Wire.h:43:7: note:   candidate expects 1 argument, 0 provided
C:\Users\zhome\Documents\Arduino\libraries\Simple_Wire\Simple_Wire.cpp: In constructor 'Simple_Wire::Simple_Wire(uint8_t)':
C:\Users\zhome\Documents\Arduino\libraries\Simple_Wire\Simple_Wire.cpp:32:41: error: no matching function for call to 'TwoWire::TwoWire()'
 Simple_Wire::Simple_Wire(uint8_t address) {
                                         ^
In file included from C:\Users\zhome\Documents\Arduino\libraries\Simple_Wire\Simple_Wire.h:28,
                 from C:\Users\zhome\Documents\Arduino\libraries\Simple_Wire\Simple_Wire.cpp:24:
C:\Users\zhome\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\Wire\src/Wire.h:73:5: note: candidate: 'TwoWire::TwoWire(uint8_t)'
     TwoWire(uint8_t bus_num);
     ^~~~~~~
C:\Users\zhome\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\Wire\src/Wire.h:73:5: note:   candidate expects 1 argument, 0 provided
C:\Users\zhome\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\Wire\src/Wire.h:43:7: note: candidate: 'constexpr TwoWire::TwoWire(const TwoWire&)'
 class TwoWire: public Stream
       ^~~~~~~
C:\Users\zhome\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\Wire\src/Wire.h:43:7: note:   candidate expects 1 argument, 0 provided
exit status 1
Error compiling for board ESP32 Dev Module.
```
I'll take a further look at it tomorrow.
Libraries Updated:
[Simple_Wire.zip](https://github.com/ZHomeSlice/Simple_MPU6050/files/9436721/Simple_Wire.zip)
[Simple_MPU6050.zip](https://github.com/ZHomeSlice/Simple_MPU6050/files/9436723/Simple_MPU6050.zip)

Test Sketch:

```
/* ============================================
  Simple_MPU6050 device library code is placed under the MIT license
  Copyright (c) 2021 Homer Creutz

  Permission is hereby granted, free of charge, to any person obtaining a copy
  of this software and associated documentation files (the "Software"), to deal
  in the Software without restriction, including without limitation the rights
  to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
  copies of the Software, and to permit persons to whom the Software is
  furnished to do so, subject to the following conditions:

  The above copyright notice and this permission notice shall be included in
  all copies or substantial portions of the Software.

  THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
  IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
  FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
  AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES, OR OTHER
  LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT, OR OTHERWISE, ARISING FROM,
  OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
  THE SOFTWARE.
  ===============================================
*/
#include "Simple_MPU6050.h"

#define MPU6050_DEFAULT_ADDRESS     0x68 // address pin low (GND), default for InvenSense evaluation board
Simple_MPU6050 mpu;

//================================================================
[... 49 lines cut ...]
{
  Serial.begin(115200);
  Serial.println(F("\nOrientation Sensor OSC output")); Serial.println();

  mpu_setup();
}
//================================================================
//===                        Main Loop                         ===
//================================================================

void loop(void)
{
  mpu.dmp_read_fifo(false); // false = no interrupt pin attachment required.
  // Tests for Data in the FIFO Buffer
  // when it finds data it runs the mpu.on_FIFO(print_Values)
  // function which we set run the print_Values Function
  // The print_Values function MUST have the following variables available to attach data
  // void print_Values (int16_t *gyro, int16_t *accel, int32_t *quat, uint32_t *timestamp)
  // Variables:
  // int16_t *gyro for the gyro values to be passed to it (The * tells the function it will be a pointer to the value)
  // int16_t *accel for the accel values to be passed to it
  // int32_t *quat for the quaternion values to be passed to it
  // uint32_t *timestamp which will be the micros()value at the time we retrieved the Newest value from the FIFO Buffer.
}
``

Wbiu, 2022-08-27:
I tried it today but unfortunatelyI still can´t compile T-T .

                                               _________("  no matching function for call to 'TwoWire::TwoWire()'  ")
                                               ▼
26 Simple_Wire::Simple_Wire() {  ←  (" no default constructor exists for class "TwoWire"C/C++(291) ")
27    begin();
28    setClock(400000); // 400kHz I2C clock. 
29 //    setWireTimeout(3000, true); //timeout value in uSec
30 }
31                                                           ▼ ---- ( " no matching function for call to 'TwoWire::TwoWire()' " )
32  Simple_Wire::Simple_Wire(uint8_t address) { ← (" no default constructor exists for class "TwoWire"C/C++(291)  ") 
33    SetAddress(address);
34 }

Log : 

Compiling .pio\build\esp-wrover-kit\src\Simple_MPU6050-master\Simple_MPU6050.cpp.o
Compiling .pio\build\esp-wrover-kit\src\Simple_MPU6050-master\Simple_Wire.cpp.o
Compiling .pio\build\esp-wrover-kit\src\main.cpp.o
Compiling .pio\build\esp-wrover-kit\lib6e7\Adafruit MPU6050\Adafruit_MPU6050.cpp.o
src/Simple_MPU6050-master/Simple_Wire.cpp: In constructor 'Simple_Wire::Simple_Wire()':
src/Simple_MPU6050-master/Simple_Wire.cpp:26:26: error: no matching function for call to 'TwoWire::TwoWire()'
 Simple_Wire::Simple_Wire() {
                          ^
In file included from src/Simple_MPU6050-master/Simple_Wire.h:28,
                 from src/Simple_MPU6050-master/Simple_Wire.cpp:24:
C:/Users/Biu/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:73:5: note: candidate: 'TwoWire::TwoWire(uint8_t)'
     TwoWire(uint8_t bus_num);
     ^~~~~~~
C:/Users/Biu/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:73:5: note:   candidate expects 1 argument, 0 provided
C:/Users/Biu/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:43:7: note: candidate: 'constexpr TwoWire::TwoWire(const TwoWire&)'
 class TwoWire: public Stream
       ^~~~~~~
C:/Users/Biu/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:43:7: note:   candidate expects 1 argument, 0 provided
src/Simple_MPU6050-master/Simple_Wire.cpp: In constructor 'Simple_Wire::Simple_Wire(uint8_t)':
src/Simple_MPU6050-master/Simple_Wire.cpp:32:41: error: no matching function for call to 'TwoWire::TwoWire()'
 Simple_Wire::Simple_Wire(uint8_t address) {
                                         ^
In file included from src/Simple_MPU6050-master/Simple_Wire.h:28,
                 from src/Simple_MPU6050-master/Simple_Wire.cpp:24:
C:/Users/Biu/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:73:5: note: candidate: 'TwoWire::TwoWire(uint8_t)'
     TwoWire(uint8_t bus_num);
     ^~~~~~~
C:/Users/Biu/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:73:5: note:   candidate expects 1 argument, 0 provided
C:/Users/Biu/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:43:7: note: candidate: 'constexpr TwoWire::TwoWire(const TwoWire&)'
 class TwoWire: public Stream
       ^~~~~~~
C:/Users/Biu/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:43:7: note:   candidate expects 1 argument, 0 provided
In file included from .pio/libdeps/esp-wrover-kit/Adafruit MPU6050/Adafruit_MPU6050.cpp:38:
.pio/libdeps/esp-wrover-kit/Adafruit MPU6050/Adafruit_MPU6050.h:21:10: fatal error: Adafruit_BusIO_Register.h: No such file or directory

*********************************************************************************
* Looking for Adafruit_BusIO_Register.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Adafruit_BusIO_Register.h"
* Web  > https://registry.platformio.org/search?q=header:Adafruit_BusIO_Register.h
*
*********************************************************************************

 #include <Adafruit_BusIO_Register.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-cpu.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-dac.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-gpio.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-i2c-slave.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-i2c.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-ledc.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-matrix.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-misc.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-psram.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-rgb-led.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-rmt.c.o
Compiling .pio\build\esp-wrover-kit\FrameworkArduino\esp32-hal-sigmadelta.c.o
*** [.pio\build\esp-wrover-kit\src\Simple_MPU6050-master\Simple_Wire.cpp.o] Error 1
*** [.pio\build\esp-wrover-kit\lib6e7\Adafruit MPU6050\Adafruit_MPU6050.cpp.o] Error 1
================================================================================================================ [FAILED] Took 7.49 seconds ================================================================================================================

ZHomeSlice, 2022-08-28:
Lets try this again
Simple_Wire is now located here: https://github.com/ZHomeSlice/Simple_Wire
Also get the latest Simple_MPU6050 here: https://github.com/ZHomeSlice/Simple_MPU6050

Compiled 100% with several ESP32 board types C3 S2 and a few others. 

Sketch uses 219226 bytes (16%) of program storage space. Maximum is 1310720 bytes.
Global variables use 10584 bytes (3%) of dynamic memory, leaving 317096 bytes for local variables. Maximum is 327680 bytes.

I hope this helps.

Wbiu, 2022-08-29:
Ohhh it compiled !!! 

I still get one question tho : 

Yaw   -8.3418,   Pitch    3.1131,   Roll    0.0821,   ax   878,   ay    18,   az 16406,   gx    -1,   gy    -1,   gz    -1

out of the three value set : the yaw pitch and roll are the one to use right ? 
I havent connect the programm that I will use it on right but later on I´ll test it out. 

Thanks for the work and help !!

ZHomeSlice, 2022-08-30:
The Yaw is calculated from the Quaternions with no gravity influence the pitch and roll have gravity influence from the accelerometer. if you would like to try gyro only change the following
```
// these are defined for you
//#define Three_Axis_Low_Power_Quaternions 3
//#define Six_Axis_Low_Power_Quaternions 6  // Default
```
replace 
`Simple_MPU6050 mpu;`
with
`Simple_MPU6050 mpu(Three_Axis_Low_Power_Quaternions );`

Try using the 3-axis Gyro only control for items that gravity is not a constant accelerating items like a rocket or falling objects.

Also, try compiling the other [Examples](https://github.com/ZHomeSlice/Simple_MPU6050/tree/master/Examples) as I believe I've resolved the custom requirements or mitigated most of the errors that were there with the latest updates.

closed 2022-09-02 by ZHomeSlice