ZHomeSlice/Simple_MPU6050 #36: It doesn't work on ESP8266 or ESP32

Olejan, 2023-06-01:
Hi!

I try to run the [Simple_MPU6050_Calibration] example (https://github.com/ZHomeSlice/Simple_MPU6050/tree/master/Examples/Simple_MPU6050_Calibration) on my ESP8266 Dev Module. It compiles. But after flashing in Serial Output, I see some kind of garbage instead of data. If I change the 
Serial Monitor speed from 115200 to 74880 baud it changes the situation. But the data in the Serial Monitor is strange:

`load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v00042050
~ld
rf cal sector: 1020
freq trace enable 0
rf[112] : 0��r��`

The situation is repeated on the ESP32, although changing the speed of the Serial Monitor does not lead to readable data.
What's wrong?

Olejan, 2023-06-01:
If I comment out all strings with "mpu" in the code from [Simple_MPU6050_Calibration](https://github.com/ZHomeSlice/Simple_MPU6050/tree/master/Examples/Simple_MPU6050_Calibration), then the Serial Monitor will display what should be in the code. But at the beginning of the message there will be some amount of garbage. If I leave the code in its original state, it will be what I described above. At the same time, my mpu6050 is definitely working, which I can verify by uploading my code.

Olejan, 2023-06-01:
[image]

ZHomeSlice, 2023-06-01:
the output looks correct you have commented out the 
  mpu.PrintActiveOffsets();
   to show the offsets that are defaulted in the mpu,
and now we are waiting for a serial input sent to the esp8266 to start the calibration.
comment out these three lines to automatically enter the calibration:

  while (Serial.available() && Serial.read()); // empty buffer
  while (!Serial.available());                 // wait for data
  while (Serial.available() && Serial.read()); // empty buffer again

Z

ZHomeSlice, 2023-06-01:
I missed your prior comment.
make sure you have the latest versions of the 
[https://github.com/ZHomeSlice/Simple_Wire](Simple_Wire)
and the Simple_MPU6050

Olejan, 2023-06-02:
Yes, I have the latest versions of the libraries and the situation has not changed.
New data - when I send any character to the Serial port, it can be seen that the program starts the sensor calibration process. But there are the same problems with displaying - unreadable characters are displayed on the Serial Monitor in one line. At the same time, the calibration process does not end - this can be seen from the filling of the monitor with symbols when it is cleared.

ZHomeSlice, 2023-06-04:
What voltage are you applying to the MPU6050
3V or 5V?
Why I ask :)
The breakout board has a 3V regulator on it and is designed to have 5V applied to the regulator. note that the connections to the ESP8266 are all at the 3V level after the voltage regulator. If you apply 3V to the regulator something like 2.3V is now at the MPU6050 and I have seen odd things happen.
[Schematic](https://components101.com/sites/default/files/inline-images/MPU6050-Circuit_0.png)
I'm struggling to determine why you would have the ESP8266 reset once the MPU becomes active. 
Unfortunately, I am not able to set up my test ESP8266 until after June 9th. I'll try to help where I can.
Z

Olejan, 2023-06-05:
Hi. Thanks a lot for your help. The voltage applied to the MPU6050 is 3.3V. But the voltage on pin 5 of Q2 of this scheme is same. I measured it.
On the other hand, I have a code with i2cdevlib library and it works fine.

Olejan, 2023-06-05:
[image]