ZHomeSlice/Simple_MPU6050 #39: Calibration doesn't work with Arduino Due

guillem13freixanet, 2023-07-18:
I've done several attempts to calibrate MPU6050 / 9250 with an Arduino Due and it never reaches the end. Code works fine with Mega 2560 though. 

Could you please help me with the Arduino Due?

Does calibration should be done everytime (sensor is done in a Quadcopter) I power up the drone or one time is just fine?

Does your library (DMP) uses accelerometer, gyroscope and magnetometer to measure and calculate pitch, roll and yaw data? or it only uses accelerometer and gyroscope data?

Thanks a lot for your help!

ZHomeSlice, 2023-07-18:
I do not have a Due to test this unfortunately. But you can use the calibration values you received from the mega to set the calibration in the Due. I will need more information about the crash to come up with a solution. 

The MPU9250 and MPU9050 have the magnetometer attached via the secondary I2C bus which is accessible from the primary I2C bus if you turn it on. This is a separate chip that is placed on the same silicone dye depending on the MPU model the manual is available as a separate document to get more details. It has a separate startup process. I've broke out the registry values in my library so that you can access them for both models of the magnetometer. The embedded digital motion processing eDMP that is loaded into the MPU has no access to the magnetometer. It's strictly uses the gyroscope and accelerometer. There is a pre-compiled motion processing library available from invecSense that integrates the magnetometer into the eDMP quaternion calculations unfortunately I have never been able to get it to compile with the Arduino ide. 
 

While the MPU 6050 is an excellent low cost architecture for accelerometer and gyro use integrating magnetometer is very difficult. 
The BNO055 chip which I have a library that I created for it is much better if you want magnetometer integration. Also note that the calibration is done internally and it is very easy to start up. 
Z

guillem13freixanet, 2023-07-18:
Thanks a lot for your answer. 

About the crash, actually there's no crash with Due. The problem is that it never reaches calibration. It keep searching for it infinitelly. 

I did test BNO055 a few months ago and it works perfectly well, but the issue I have is that I need fast output sample rate, at least 200Hz, which is what I get with your library and MPU6050. With the BNO055 I couldn't get such fast sample rate and so it was not suitable to controll a drone. 

What I would like to do, at least (coming back to the MPU6050 topic), is to measure absolut pitch and roll at the start up, in order to avoid any offset. Your proposed calibration is very depending on how flat is the surface is... and I can hardly find "perfect" surface to calibrate it everytime. That's why I thought of using the magnetometer to find this 0 offset at the begining. Do you have an idea of how to do it? I appreciate any idea :-)

ZHomeSlice, 2023-07-18:
Once the accelerometers are set the flat surface is not an issue. Did gyroscope if it's calibrated close will be enough as my code has the automatic gyro calibration routine enabled in the eDMP. So if you use accurate offsets then you should be close enough. The six degrees of freedom uses gyro and accelerometer to calibrate level. As long as it knows what level is. This is done by setting the MPU6050 on a flat surface and running the calibration code. The offsets this generate should be used every time and the initial starting point for this specific MPU6050. The only drift comes from the yaw access. Calibration for this does not require a level surface it just requires no motion. And again the embedded DMP firmware has an auto calibration for the gyro internally and after a few moments of stability it recognizes this and will zero out the offsets further within the embedded DMP firmware. 
Hope this helps.
Z